makerbase-mks/MKS-DLC32 #431: Cant print straight raster images (always wavy in x (raster) direction)

zuber6, 2023-11-04:
Tried many things. Here is what I get with my ramps board using grbl mega 5x:

[image]

Using identical parameters this is what I get with the dlc32:

[image]

If i take the identical picture without fill, again in raster mode I get

[image]

Not perfect, but OK. This is in vector mode with the dlc32:

[image]

No issues.

Help! Have had to go back to the ramps setup. This is on a diy machine adapted from a CNC. The only thing I can think of is that it has a screw drive on x and the 200 steps per mm is overpowering the dlc32 (lol) compared to the mega 3560 which makes absolutely no sense at all.

zuber6, 2023-11-20:
Hello? Anyone from mks monitoring this forum?

BaBaika666, 2023-12-14:
Здравствуйте. Вы нашли решение данной проблемы?

zuber6, 2023-12-14:
No. Still waiting for mks to respond.

calfizzi, 2023-12-25:
Hi [zuber6](https://github.com/zuber6) I Found the Same Issue; what I understood is: this behavior happen when there is big difference between axes speed of command "G0" and "G1". the G0 command works (I suppose) with the half of Maximum Speed ($110, $111,$112), instead, the G1 Command is regulated by the engraving speed you are using....
...for the moment to reduce the effect of this issue you can temporary change the maximum speed (into the parameter setting) to the double of your engraving speed so "G0" and "G1" will work with the same speed...
I think is a Bug but there is no answer yet by the developing team! 

regards,
Mario

zuber6, 2023-12-30:
Thanks calfizzi! Unfortunately I already had the max speed set to 2x the engraving speed (not intentionally). However, by setting the max speed = engraving speed I saw a significant improvement. Still nowhere near the results from a ramps board but much better. This may be because I am using a screw drive on the x-axis requiring a larger number of steps per mm over a belt drive.
Thank you so much for your input!

zuber6, 2024-03-01:
Solution is in another post. $0=10 and $1=5. Still think this is a probelm with the firmware though.