makerbase-mks/MKS-DLC32 #439: LS ESP32 Pro V 2.1

Plottmania, 2023-11-22:
Könnte man mal ein paar Infofmationen bekommen bezüglich der $?=?

Auf Verbindung warten...

Auf Verbindung warten...

H���-�ok

ok

[V4.00(8M.H35.20230313)]

[VER:1.1h.2023031301:]

[OPT:VMPH,63,256]

Target buffer size found

[MSG:Using machine:MKS DLC32]

[MSG:No Wifi]

ok

$$

**$54=12345678
$53=MKS_DLC
$52=12345678
$51=MKS_DLC**
$0=10
$1=25
$2=0
$3=5
$4=0
$5=1
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=0
$23=5
$24=300.000
$25=1000.000
$26=250.000
$27=1.000
$28=1000.000
$30=1000.000
$31=0.000
$32=1
**$41=1
$47=1
$46=1**
$102=100
$101=80
**[image]**
$100=80.000
$103=100.000
$104=100.000
$105=100.000
$110=6000.000
$111=6000.000
$112=6000.000
$113=1000.000
$114=1000.000
$115=1000.000
$120=500.000
$121=500.000
$122=500.000
$123=200.000
$124=200.000
$125=200.000
$130=450.000
$131=450.000
$132=50.000
$133=300.000
$134=300.000
$135=300.000

ok

Venom311085, 2024-01-11:
Hallo weißt du wo ich die Firmware herbekomme ?

Plottmania, 2024-01-12:
Ja, aber aktuell hab ich auch nur V. 4.1 und nicht wie hier die 4.0 über GitHub bekommen.

Plottmania, 2024-01-12:
Bitte....

Venom311085, 2024-01-12:
Boa Hammer du bist meine Rettung wenn es geht ich wollte meinen Gravierer ins WLAN einbinden und habe die falsche Firmware geflasht und er steht still. Hoffe es  klappt mit dieser. Teste es nachher

MJ666, 2024-01-23:
Ich habe die Original V4.00(8M.H35.20230313) Firmware mal gesichert falls die jemand wirklich braucht. Die neuste Version vom Datum ist wohl die von TT V4.00(8M.H35.20231120). Diese ist immer aber immer noch V4.0? wis nicht ob TT die Firmware nun unabhängig von MKS pflegt. Habe auch die Firmware vom zugehörigen Display gesichert ESP_TFT35 v2.0_20231215.

Venom311085, 2024-01-24:
> Ich habe die Original V4.00(8M.H35.20230313) Firmware mal gesichert falls die jemand wirklich braucht. Die neuste Version vom Datum ist wohl die von TT V4.00(8M.H35.20231120). Diese ist immer aber [...]

Ja ich hätte sie gerne wenn das geht. 
Welches display hast du ? Ich suche noch eins für meinen laser. Ich habe aber nur diesen komischen Display Port oben auf dem board

MJ666, 2024-01-24:
Ich habe dieses Display:

https://www.[twotrees3dofficial.com/de/products/two-trees-touch-screen-for-tts-series-laser-engraver-tts-25-tts-55-tts10?shpxid=affd6656-14ce-4642-b1f8-784a1cafea68](https://www.twotrees3dofficial.com/de/products/two-trees-touch-screen-for-tts-series-laser-engraver-tts-25-tts-55-tts10?shpxid=affd6656-14ce-4642-b1f8-784a1cafea68)

Für die Firmware kannst du mir deine E-Mail Adresse zukommen lasen. Dann schicke ich sie Dir.

kaiatkins, 2024-01-29:
https://drive.google.com/drive/folders/1ImJkcrlZymhxi5SH7pRYX1kzV-vIZiA2?spm=a2g0s.imconversation.0.0.38ce3e5fwJNQ2m

LAS_MKS_ESPS3 PRO V4.1_V1.0.0_20240103_Bate.bin

zombodotcom, 2025-04-07:
Look what I found

```
V4.00(8M.H35.20240226)]
[VER:1.1h.2024022601:]
[OPT:VPH,63,256]
[MSG:Using machine:MKS DLC32]
ok
[MSG:Mode=AP:SSDI=MKS_DLC12457:IP=192.168.4.1:MAC=xxxxxxxxxxxxxxxxxxx
ok
ok
ok
[MSG:Wifi Connecting...]
[MSG:Wifi Connected]
ok
[MSG:Start mDNS with hostname:http://mks_grbl.local/]
[MSG:HTTP Started]
[MSG:TELNET Started 8080]
ok
[MSG:Wifi Connecting...]
[MSG:Wifi Connected]
ok
[MSG:Start mDNS with hostname:http://mks_grbl.local/]
[MSG:HTTP Started]
[MSG:TELNET Started 8080]
192.168.0.78
ok
_connected=1
_connected=1
_connected=1
_connected=1
_connected=1
_connected=1
```

[image]

[image]

https://baizhongyun.cn/home/mkstoolview

works for me!

also used the mks tool in the firmware folder here in this github repo