makerbase-mks/MKS-DLC32 #440: Z Probe is not reacting at all.

ElectricBluefishProductionsInc, 2023-11-23:
I purchased a z-axis probe for the Makerbase DLC32 v2.1 board:
t.ly/3bFiv

I wired it like this:
[image]

Running on a 24volt power supply.

My default firmware V2.10(8M.H35.20200621) W T:35config is:
$$
$0=10
$1=25
$2=0
$3=6
$4=0
$5=1
$6=1
$10=1
$11=0.010
$12=0.002
$13=0
$20=1
$21=1
$22=1
$23=3
$24=300.000
$25=2500.000
$26=250.000
$27=1.000
$28=1000.000
$30=1000.000
$31=0.000
$32=1
$38=1
$40=1
$100=80.000
$101=80.000
$102=80.000
$103=100.000
$104=100.000
$105=100.000
$110=9000.000
$111=9000.000
$112=3000.000
$113=1000.000
$114=1000.000
$115=1000.000
$120=500.000
$121=500.000
$122=500.000
$123=200.000
$124=200.000
$125=200.000
$130=1574.000
$131=1524.000
$132=250.000
$133=300.000
$134=300.000
$135=300.000

If I change $6 = 0

and run: G38.2 Z-200 F500
ALARM:4
Probe fail. The probe is not in the expected initial state before starting probe cycle, where G38.2 and G38.3 is not triggered and G38.4 and G38.5 is triggered.

Set $6 = 1 again and I get:
G38.2 Z-200 F500
[MSG:Found]
ALARM:5
Probe fail. Probe did not contact the workpiece within the programmed travel for G38.2 and G38.4.
[PRB:0.000,0.000,0.000:0]
ok

With this last command the probe does settle on the piece but the system does not set it and continues to depress the plunger until I stop it well after the point it should have tripped the sensor.

What don't I understand here? Why would this not work.

ElectricBluefishProductionsInc, 2023-11-23:
I reconnected the wire plug ends that plug into the board to be sure there was no problem with the connection. Now I am getting:
G38.2 Z-200 F500

[MSG:Found]
[PRB:1.000,1.000,-33.862:1]
ok
[MSG:Flame err!Please restart the machine]
ALARM:3
Reset while in motion. Grbl cannot guarantee position. Lost steps are likely. Re-homing is highly recommended.
Grbl 1.1h ['$' for help]
[MSG:'$H'|'$X' to unlock]
[MSG:Caution: Unlocked]
ok
[MSG:Flame err!Please restart the machine]
Grbl 1.1h ['$' for help]
[MSG:'$H'|'$X' to unlock]
[MSG:Caution: Unlocked]
ok

ElectricBluefishProductionsInc, 2023-11-23:
Actually the [MSG:Flame err!Please restart the machine] is not reappearing after I restarted the machine so it is likely unrelated.

alhaddar, 2023-11-28:
change $6=1 to $6=0

naudevisser28, 2025-05-06:
Hi
I have the same problem
Morning.  Do you help with technical info on why I cannot get the z probe zero to work 
I have a  3018 Pro CNC Carving Machine.
Im using a  DLC32 V2.1   Controller board  and Universal Gcode Sender 20250108
Firmware:  grbl 1.1h probe makerbase dlc32 v2
I can not get it to do the z probe to determine Z0
Im using Gnd and S on the probe connector
It gives error or try to push the probe through the pcb
I will be very happy if you can maybe tell me what the problem can be.