makerbase-mks/MKS-DLC32 #444: mks dlc32 kendiğinden resetlemesi

hcenqiz, 2023-12-11:
merhaba mks dl32 v2 kartım çalışma esnasında kapanıp açılıyor . sebebini çözemedim. lightburn kullanıyorum.
V2.0.3_H35_20211108_N.bin
V2.0.4_H35_20211123_N.bin
V2.0.6_H35_20211207_N.bin
V2.0.7_H35_20211213_N.bin
V2.0.8_H35_20211222_N.bin
V2.0.8_H35_20220105_N.bin
V2.10_H35_2022_0621_N_ZX_001.bin
V2.20_H35_2022_1101_N_ZX_002.bin
V2.30_H35_2022_1223_N_ZX_001.bin
bu firmware denedim sonuç yine aynı oluyor.

hcenqiz, 2023-12-11:
> merhaba mks dl32 v2 kartım çalışma esnasında kapanıp açılıyor . sebebini çözemedim. lightburn kullanıyorum. V2.0.3_H35_20211108_N.bin V2.0.4_H35_20211123_N.bin V2.0.6_H35_20211207_N.bin [...]

Akış başlıyor
Layer C00
Bağlantı koptu
Akış tamamlandı 5:19
ok
[PRODUCER:MKS DLC32]
[V2.30(8M.H35.20221223)]
[VER:1.1h.2022122301:]
[OPT:VMPH,63,500]
Target buffer size found
[MSG:Using machine:MKS DLC32]
[MSG:No Wifi]
ok
işlem yarıda kesiliyor nedeni nedir?

gokmenhakan, 2024-03-08:
Sorunu çözebildiniz mi?  MKS LS ESP32 PRO GRBL aldım. Bu kartda da aynı sorunu yaşıyorum.
12v luk lazerim var. Güç %50 altlarında oldu mu sorun yok. Ama &%50 üzerine çıkınca connection lost hatası oluşuyor.

hcenqiz, 2024-03-08:
Evet sorun iki sebep kaynaklı

1. Sorun kablo dan kaynaklı
  Çözüm: kablo kesitinin miktari artirildi ve ek kesinlikle yok.

2. Sorun aşırı akım cekmesi durumu anahtarda deformasyona neden olmuş.
   Çözüm: Anahtar degistirildi.

Not: lazer akımı yüksek olduğunda kartda voltaj akım dalgalanması oluyor.
mks kartindan yalnizca pwm sinyali alindi. Lazer beslemesini direk güç
kaynağından yapıldı.

Umarım sorunu çözersiniz.

gokmenhakan, 2024-03-08:
Teşekkürler.  12v 5A lik güç kaynağı alıp bununla kullanmaya çalıştım. 
Ben başka cnc shield ile aynı lazeri kullanınca hiç bir sorun yaşamıyorum.
Bana sorun psu  dan kaynaklanabileceğini söyledi söyledi ali express satıcısı ama testlerim bunu çürütüyor. 3 ayrı güç kaynağı denedim ve 2 sinde bu sorun var 12V da sorunlu çalışıyor. Güç kaynaklarından birisi akımı gösteriyor. diğerinin yani 3. Güç kaynağının voltajı yüksek 19v 3.4 amper. Onda sorun yok ama lazere 19 v u aldığı gibi gönderiyor.  Bu da lazerin 12v 3A lik max girişinden fazla olduğundan zarar vereceğini düşündüğüm için çalıştırmıyorum. Sadecw 1 kaç test aldım boşta bile çok Yüksek çalışıyor fanlardan belli

hcenqiz, 2024-03-08:
15watlik lazerim var . Lazer akimi olcumde 1.8amper cikti. psu12 volt 15
amper.
 Ben sorunu böylece cözdüm ,size yazdiklarimi yaptigimda.
Suan ki sorunum ;
Gerekli ayarlari yapinca step motorlara hız veremiyorum. Baska bir ayarda
calistirinca hız artiyor lakin çizim kötü oluyor. Lightburn kullaniyorum
lisanslı

gokmenhakan, 2024-03-08:
Bendeki lazer 5.5w lık sculpfun s9 lazer modülü. Küçük işler için. Bu voltaj  ve akım olayı konusunda mks çok nazlı sanırım.  Sizdeki ve bendeki 12volt lazer bu kadar etkilememeli aslında.
Ben de lightburn de test ediyorum şuan.

hcenqiz, 2024-03-08:
Mks kart kullanim açısından iyi fakat kötü yanları var.. benbox programina
izin yok

gokmenhakan, 2024-03-08:
Siz lazeri harici güç kaynağına bağladığınızda sadece pwm i mks lazer kartından sağlayıp voltaj  ve gnd yi hariciden lazerin ilgili portlarına bağladınız değil mi?  Bu yöntemi deneyince bende pwm değişmiyor yani yazılımdan lazerin gücünü ayarlayamıyorum bağlantı etiketlerinde de sıkıntı var bunların. Boardun ön orta ucu ttl i gösterirken arka tarafında ttl etiketi ortada değil voltaja göre 3. Uçta.

hcenqiz, 2024-03-08:
Gnd ve Pwm  board dan
+12volt güç kaynagindan

gokmenhakan, 2024-03-11:
Yeni bir güç kaynağı siparişi verdim. Güç kaynağı ile ilgili şüpheleri aklımdan silmeye çalışıyorum malesef.
Son mesajınızı dikkate alacağım. Pwm ve Gnd yi yanlış etiketledikleri için bir de ters bağlayıp denemeye çalışacağım harici olarak.

hcenqiz, 2024-03-11:
Sakin pwm ile gnd yi ters bağlamayin.

gokmenhakan, 2024-03-12:
Teşekkür ederim. Desteğiniz için ters bağlama fikrinden sizin uyarınız ile vazgeçtim.
Evdeki monitörümün adaptörü 12v 3A miş onunla denedim 1 kere hata verdi. Stabilitesini denemek lazım. Güç kaynakları sağlam olmalı sma bu kartlar da çok hassas sanırım. Diğer board sıkıntısız çalışırken mks nin böyle davranması çok farklı.
Umarum sizde sorunlarınızı çözersiniz.
Arduino uno ve 80 tl lik  cnc shield ile gayet güzel çalışıyor. Aslında.

hcenqiz, 2024-03-12:
Tesekkur ederim. Size tavsiyem 10 amper güç kaynagindan az olmasin. Kolay
gelsin iyi calismalar

gokmenhakan, 2024-03-12:
Tavsiyenizi dikkate almaya çalışacağım. Şuan sabahtan beri yaptığım testlerde 12V 3A ile %100 güçte sorun yaşatmadı.
Bu da demek oluyor ki kaliteli güç kaynağı çok önemli. Şuan için benim lazerim 5.5w olduğu için 5A lik yeter diye düşünüyorum ama  ne olup olmayacağı hiç belli olmuyor. Temkinli olmak lazım. Çok teşekkür ederim size de iyi çalışmalar.

iskostak, 2025-03-23:
Merhaba,  WHY DOES MKS DLC32 V2 CNC X-Y AXIS CROSS?  Niye X-Y ekseni çapraz çalışıyor çözemedim ?

gokmenhakan, 2025-03-23:
> Merhaba, WHY DOES MKS DLC32 V2 CNC X-Y AXIS CROSS? Niye X-Y ekseni çapraz çalışıyor çözemedim ?

Hi you can change it in machine settings.

iskostak, 2025-04-19:
I just bought a second Makerbase MKS DLC32 V2.1 board, the same thing happens there, while this y button works both x and y axis, the x axis only works in one direction, I also changed the motor drivers, my computer is W10, 64 bit. I loaded different firmwares. The situation is the same. I still can't figure out the reason?

Yeni ikinci bir Makerbase MKS DLC32 V2.1 kart daha aldım, onda da durum aynı, bu y butonu ile hem x hem y eksenini çalıştırırken, x ekseni sadece tek yönde çalışıyor, motor sürücülerinide değiştirdim, bilgisayarım W10, 64 bit. Değişik firmware ler yükledim. Durum aynı. Sebebini hala çözemedim ?

iskostak, 2025-04-19:
https://github.com/user-attachments/assets/8d2c6467-7bfb-491a-a82c-9a229e86b680

iskostak, 2025-04-20:
OK, I solved the problem with the right software.
However, the installation videos of the Makerbase Mks Dlc32 card are very incomplete and insufficient. 
Although I am a mechanical engineer, I tried very hard to find the right software. It is more difficult for amateurs!