makerbase-mks/MKS-DLC32 #448: MKS DLC 32 Limit Switch connection

esloch10, 2024-01-13:
Hello,

I'm seeking guidance on connecting a limit switch to the MKS DLC32 board and would appreciate your assistance with the correct procedure.

Initially, my plan was to use a simple two-wire switch, which I believe should function adequately. However, my research indicates that the MKS DLC32 board is designed for limit switch modules with three wires.

Considering I hadn't initially planned for modules, I'm contemplating a DIY solution based on a diagram I found. However, I'm unsure of its accuracy and would appreciate your insights.

Additionally, I intended to integrate an emergency stop button into my setup but haven't identified a dedicated connector for it. What course of action would you recommend?

Thank you for your assistance.

Jennywhite2, 2024-01-15:
[image]

Jennywhite2, 2024-01-15:
[image]

Jennywhite2, 2024-01-15:
[image]

Jennywhite2, 2024-01-15:
[image]