makerbase-mks/MKS-DLC32 #453: FIRMWARE 8M.H35.20230527 ????  WHERE IT IS?

klingklang01, 2024-01-22:
I just got a new MKS DLC32 board and the firmware on it is: 8M.H35.20230527.  I don't see it here. Is there another place for firmware or it because MKS too lazy to update their firmware pages?  I wish to see if there updates to this one or at least get a genuine backup of this one.

MJ666, 2024-01-23:
You can backup your firmware yourself with ESPTool: 

esptool --port COMx --baud 115200 read_flash 0x000000 0x800000 firmware.bin

klingklang01, 2024-01-23:
Thank you , I found that yesterday.   "Unfortunately we don't expect any updates from MKS here."  Why? any idea? This not suppose to be the official place for firmware update?  Then how to know if there is a new one?

MJ666, 2024-01-24:
This should be the right place for updates but there is no activity by MKS since quote some time.

reset12, 2024-02-17:
Hello Klinklang01,
can you send me your firmware because my is older. My email is reset12@gmx.de

MJ666, 2024-02-18:
8M.H35.20230527 is only distributed on recent MKS32 boards. It was not published anywhere else yet. One option would be to get someone with this firmware to make a backup as described above. Anyhow it looks to be there are issues with this version and the M4 command: https://forum.lightburnsoftware.com/t/another-m4-error-with-dlc32-v2-1-003/109672

kris0725pl, 2024-03-07:
I have dumped firmware with .cn file Frombork longer ray5 20W.

mellbo, 2024-06-25:
[fw_8M.35.20231206.zip](https://github.com/user-attachments/files/15971907/fw_8M.35.20231206.zip)

schneidervs, 2025-12-05:
[firmware_MKS_DLC32_V2.1(8M.H35.20241121).bin.zip](https://github.com/user-attachments/files/23956948/firmware_MKS_DLC32_V2.1.8M.H35.20241121.bin.zip) for esptool