makerbase-mks/MKS-DLC32 #454: dls 32 with 2 external drivers and separate power supplies.

Strummerone, 2024-01-25:
Hi I'm using DLC 32 on a DIY CNC with 2 external drivers on the Y axis. As I'm using external drivers  Ive connected both sets of control wiring from the board to each driver via the set of pins marked Y axis (Dir,S, En etc). Problem is that if I have just one driver connected everythings fine, however when both drivers are connected, one of them (the drivers) go into fault mode( Steady red light). I read somewhere that both driver have to be on the same power supply, mine have separate 36v power supplies (Cheaper believe it or not) Is this true or am I missing something? Anyone any ideas..

MJ666, 2024-01-26:
As long you have a common ground for your the DLC32 and the 2 external drivers there is no reason why it should not work. Can you post a schematic of your build or may be picture how it is connected?

Strummerone, 2024-01-28:
[As installed wiring.pdf](https://github.com/makerbase-mks/MKS-DLC32/files/14076996/As.installed.wiring.pdf)

I,ve been tinkering and managed to get both stepper motors on the Y axis working from the one power supply. Only trouble is that when I connect further axis i.e X or Z neither of them want to work with the Y but if I set them up on them individually `9`without the Y connected they work.

MJ666, 2024-01-29:
How is the  DLC32 powerd? I guess it is another power supply? You may need a common ground of all power supplys with strong wirering. It should not rely on the thin ground traces of the DLC32 itself only. This may lead to potential shift on your ground and causing your problems.

Strummerone, 2024-01-29:
[As Installed with ground.pdf](https://github.com/makerbase-mks/MKS-DLC32/files/14087340/As.Installed.with.ground.pdf)

Please see new drawing with additional bits. Would adding the interconnecting ground to each driver (see drawing) help?
I could cross bond (British expression for equipotential bonding) the power supply chassis to each other with a larger earth cable but I think the chassis is already earthed via the incoming supply. 
I can't get my head around why these won't work together as the DLC32 is just sending out signals. Oh and by the way, the power supply and the supply for the DLC32 are from the same source.

Luis-Reis100, 2024-10-17:
> Dear [Strummerone](https://github.com/Strummerone) , have you managed to solve the issue, having two drivers at only one DLC32 Y axis output?

Strummerone, 2024-10-21:
My apologies for not communicating sooner. Yes I solved the 2 motor issue. My fault, I hadn't noticed that 2 of the connections got crossed. I am now looking at a way to control the drivers via the board (firmware) using another axis and will try to add some sort of safety into the equation in case one motor works and the other doesn't leading to catastrophe. Many thanks for all the help..

Luis-Reis100, 2024-10-21:
Are you using the FluidNC as a firmware to the MKSDLc32 or the original
GRBL firmware that comes with the board?
I`ve read that Fluidnc allows you to better manage what you want for any
axis output to perform.  It seems that the output & input designation with
the FluidNc is more flexible.
I really don't have any experience with FluidNC, I've just read it.

Strummerone, 2024-10-22:
Hi there.

Yes, I’m using Fluidic on a Makerbase DLC32 over wifi.
Fluidnc is quite flexible, however I don't use the web interface much. The LCDs that come with the dlc32 won’t
 work once Fluidnc is loaded. I mostly use UGS to send Gcode to the machine which works fine.
At the moment I haven't the time to look further into how to set up the dual motor control via Fluidic, but I will eventually get round to it,  I don't like the idea that one day one of the motors dies leaving the other to run amuck.  I will need to work out how to practically set up something like motion sensing or current flow sensing to detect if both motors are running, if not go to alarm.
Kind regards
Pete

Luis-Reis100, 2024-10-22:
Pete,

Your concern about this issue of having one of the motors getting down is
important, would severely damage the mechanical parts of the machine.

If you don't really mind, when you find a way to solve this issue, i'd ask
you to  share with me.

Regards,
Luis Reis

Strummerone, 2024-10-22:
Hi again.
No problem sharing however it might be a while as I’m up to my eyes in it at the moment.
Regards
Pete

Kind Regards