makerbase-mks/MKS-DLC32 #455: New Chalenge for Atomstack A10 v2 Firmware Upgrade for AirAssist

GrosBerto, 2024-02-04:
Hello,

On my **Atomstack A10 V2**, I would like update/upgrade the firmware for implement air assit.
I think mainboard can to do. Because "**Air Assit**" is write on this one. But i think the original firmware is locked by Atomstack.
When i test **M8** or M7 for active air assit with Light-burn, **nothing change**. Output voltage of air assit pin **stay at 12V**.
Main board is **LS ESP32 V1.0** with **M17901** write on back. You can see mainboard bellow.

I know Marlin mainboard i have already modify and make some 3D printer, make a CNC milling with Marlin mainboard. But this environment this new for me. 

I found nothing similar for my update on web. Someone could help for this topic ? Some Ideas ?

Thanks in advance.

[image]

[image]

AreyouShura, 2024-04-02:
Hi there, i had the same problem and I went by the other way. a brougth a 12v TTL relay module and connected it to the laser ttl output pin. so i have my air pump turned on when the laser is on. when i dont need an air assist i switch the relay off. Rude but it works.

GrosBerto, 2024-04-06:
Hello, nice it's a good idea. I'm feel less alone with my trouble. But community is poor for this machine. I will think about your solution and maybe i do this. Or change motherboard, i will see.Thanks

DiogoMonteiro1992, 2024-08-19:
Hello,
Sorry, I know this is not related to the topic, but could you send the original firmware for this board?
I've already tried through support, but the firmware they send doesn't always work well.
It's for an Atomstack A5 V2.

Thank you so much if you can share it.

AreyouShura, 2024-08-19:
Hello,I don't know how to download the firmware from machine.

ZioMauri57, 2025-01-04:
You could make a separate circuit, connect it to the tx pin of the cp210 and capture all the gcode, when the code 07 or 08 passes, the pump is turned on. Do you think this is reasonable?

GrosBerto, 2025-01-04:
> You could make a separate circuit, connect it to the tx pin of the cp210 and capture all the gcode, when the code 07 or 08 passes, the pump is turned on. Do you think this is reasonable?

Hello, i think it's a good idea. I need to investigate a solution to listen the tx signal.

ZioMauri57, 2025-01-04:
> > You could make a separate circuit, connect it to the tx pin of the cp210 and capture all the gcode, when the code 07 or 08 passes, the pump is turned on. Do you think this is reasonable?
> 
> Hello, i think it's a good idea. I need to investigate a solution to listen the tx signal.

It should be fairly simple, using an arduino to read the data, process it, and turn on the pump via a relay.