makerbase-mks/MKS-DLC32 #457: Display firmware is not running on TTS brandend TTF35 display

MJ666, 2024-02-28:
The firmware https://github.com/makerbase-mks/MKS-DLC32/blob/main/MKS-ESP32-PRO-main/TFT35-firmware/ESP_TFT35_V4.1_2023082201.bin is not working on the TTS display. The screen is showing garbage. I don't tried the other two published firmware builds. I reverted back to the backup i did before the firmware update. Anyhow the firmware on this display looks to be even newer the the latest published firmware? The date code says ESP_TFT35 v2.0_20231215. The version looks to be behind but date is newer? This display is working on the MKS LS ESP32 V2.1.

Staszek-org, 2024-04-14:
> The firmware https://github.com/makerbase-mks/MKS-DLC32/blob/main/MKS-ESP32-PRO-main/TFT35-firmware/ESP_TFT35_V4.1_2023082201.bin is not working on the TTS display. The screen is showing garbage. I [...]

I do the same, but I didn't make backup of the working firmware. Can you send me link to your backup firmware, that is working, please?

MJ666, 2024-04-15:
Please join Discord for private conversation and send me your e-mail. https://discord.gg/BqjUESKq

Charus-OV, 2024-05-11:
> Пожалуйста, присоединяйтесь к Discord для приватного разговора и отправьте мне свой e-mail. https://discord.gg/BqjUESKq

Hello friend. Could you send me the firmware by email [4arusov75@gmail.com](mailto:4arusov75@gmail.com)

cristianodems, 2024-05-22:
> I also didn't make the backup and TTS doesn't answer me either, can you send me the old firmware too please?

I also didn't make the backup and TTS doesn't answer me either, can you send me the old firmware too please?

MJ666, 2024-05-23:
> > I also didn't make the backup and TTS doesn't answer me either, can you send me the old firmware too please?
> 
> I also didn't make the backup and TTS doesn't answer me either, can you send me the old firmware too please?

Please join Discord for private conversation and send me your e-mail address. https://discord.gg/BqjUESKq

cristianodems, 2024-05-23:
> Please join Discord for private conversation and send me your e-mail address. https://discord.gg/BqjUESKq

The invite is invalid.

MJ666, 2024-05-23:
https://discord.gg/3shBsh3e

Charus-OV, 2024-06-11:
I have compiled a new firmware

MJ666, 2024-06-11:
Is there any source code for the ESP TTF35 display firmware?

Grex3D, 2024-08-09:
Hello, if anyone can send me the original firmware (backup), I would really appreciate it.
I've been trying to contact MKS with no response.
I'm completely stuck. :(

Thanks

alarshi63, 2025-01-23:
Can you share the backup you have with me?

MJ666, 2025-01-23:
This should work: https://github.com/makerbase-mks/MKS-DLC32/blob/main/MKS-ESP32-PRO-main/tft35%20firmware