makerbase-mks/MKS-DLC32 #461: LS ESP32 Pro V 2.1 - CORE XY

bobricius, 2024-03-09:
Is possible use this board with CORE-XY setup, How to configure ?

kingjia1989, 2024-06-23:
have you find the firmware？

Golabstech, 2024-10-27:
Hello, there seems to be a corexy configuration in this .rar file. Can you try to upgrade the firmware on your SD card?

MKS-ESP32-PRO-main/hardware/LS ESP32 PRO V2.1_002/firmware.rar

zombodotcom, 2025-04-07:
Look what I found

```
V4.00(8M.H35.20240226)]
[VER:1.1h.2024022601:]
[OPT:VPH,63,256]
[MSG:Using machine:MKS DLC32]
ok
[MSG:Mode=AP:SSDI=MKS_DLC12457:IP=192.168.4.1:MAC=xxxxxxxxxxxxxxxxxxx
ok
ok
ok
[MSG:Wifi Connecting...]
[MSG:Wifi Connected]
ok
[MSG:Start mDNS with hostname:http://mks_grbl.local/]
[MSG:HTTP Started]
[MSG:TELNET Started 8080]
ok
[MSG:Wifi Connecting...]
[MSG:Wifi Connected]
ok
[MSG:Start mDNS with hostname:http://mks_grbl.local/]
[MSG:HTTP Started]
[MSG:TELNET Started 8080]
192.168.0.78
ok
_connected=1
_connected=1
_connected=1
_connected=1
_connected=1
_connected=1
```

[image]

[image]

https://baizhongyun.cn/home/mkstoolview

works for me!

also used the mks tool in the firmware folder here in this github repo