makerbase-mks/MKS-DLC32 #462: Flash MakerBase MKS DLC32 v2.1

carlosmorenoaguilera, 2024-03-11:
Hi i need some assistance 

im trying to flash this board for a CNC router:

MakerBase MKS DLC32 v2.1 

when im trying to flash the process end without error (MKS Laser Tool V2.1.1):

[image]

this is the log 

`C:\[user]\GitHub\MKS-DLC32\MKS-DLC32-main\firmware\TS35\Laser\Normal\Board_V2.0\V2.0.8_H35_20220105_N.bin
0

Panel-0Command: --port COM5 --baud 115200 --after no_reset --chip esp32 write_flash --flash_size 8MB --flash_mode dout 0x00 C:\[user]\GitHub\MKS-DLC32\MKS-DLC32-main\firmware\TS35\Laser\Normal\Board_V2.0\V2.0.8_H35_20220105_N.bin

Running --port COM5 --baud 115200 --after no_reset --chip esp32 write_flash --flash_size 8MB --flash_mode dout 0x00 C:\[user] \GitHub\MKS-DLC32\MKS-DLC32-main\firmware\TS35\Laser\Normal\Board_V2.0\V2.0.8_H35_20220105_N.bin...
esptool.py v4.4-dev
Serial port COM5
Panel-0 Connecting...
......
Chip is ESP32-D0WD (revision v1.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
Panel-0 MAC: 08:d1:f9:4d:68:a8
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x003fffff...
Compressed 4194304 bytes to 1099450...

Panel-0 Writing at 0x00000000... (1 %)
Panel-0 Writing at 0x00012438... (2 %)
Panel-0 Writing at 0x00021d54... (4 %)
Panel-0 Writing at 0x0004cbc1... (5 %)
Panel-0 Writing at 0x00098928... (7 %)
Panel-0 Writing at 0x000c10de... (8 %)
Panel-0 Writing at 0x000e03c2... (10 %)
Panel-0 Writing at 0x000ea6fa... (11 %)
Panel-0 Writing at 0x000f5278... (13 %)
Panel-0 Writing at 0x000ff3e0... (14 %)
Panel-0 Writing at 0x001100d6... (16 %)
Panel-0 Writing at 0x0011ce99... (17 %)
Panel-0 Writing at 0x001362e7... (19 %)
Panel-0 Writing at 0x0013c271... (20 %)
Panel-0 Writing at 0x001424d3... (22 %)
Panel-0 Writing at 0x00147e2b... (23 %)
Panel-0 Writing at 0x0014df27... (25 %)
Panel-0 Writing at 0x00154100... (26 %)
Panel-0 Writing at 0x0015abad... (27 %)
Panel-0 Writing at 0x00160800... (29 %)
Panel-0 Writing at 0x00166714... (30 %)
Panel-0 Writing at 0x0016bcf3... (32 %)
Panel-0 Writing at 0x0017137c... (33 %)
Panel-0 Writing at 0x00176784... (35 %)
Panel-0 Writing at 0x0017c5c2... (36 %)
Panel-0 Writing at 0x0018240f... (38 %)
Panel-0 Writing at 0x00187700... (39 %)
Panel-0 Writing at 0x0018c8e3... (41 %)
Panel-0 Writing at 0x00191e14... (42 %)
Panel-0 Writing at 0x001973ee... (44 %)
Panel-0 Writing at 0x0019c7d0... (45 %)
Panel-0 Writing at 0x001a3a9d... (47 %)
Panel-0 Writing at 0x001abde2... (48 %)
Panel-0 Writing at 0x001b1628... (50 %)
Panel-0 Writing at 0x001b6e61... (51 %)
Panel-0 Writing at 0x001bc7e2... (52 %)
Panel-0 Writing at 0x001c26e2... (54 %)
Panel-0 Writing at 0x001c7fb0... (55 %)
Panel-0 Writing at 0x001cda31... (57 %)
Panel-0 Writing at 0x001d3347... (58 %)
Panel-0 Writing at 0x001d906d... (60 %)
Panel-0 Writing at 0x001dec14... (61 %)
Panel-0 Writing at 0x001e5810... (63 %)
Panel-0 Writing at 0x001eb2ff... (64 %)
Panel-0 Writing at 0x001f1158... (66 %)
Panel-0 Writing at 0x001f6ebc... (67 %)
Panel-0 Writing at 0x001fca36... (69 %)
Panel-0 Writing at 0x0020281a... (70 %)
Panel-0 Writing at 0x00208626... (72 %)
Panel-0 Writing at 0x0020e173... (73 %)
Panel-0 Writing at 0x00214158... (75 %)
Panel-0 Writing at 0x0021a964... (76 %)
Panel-0 Writing at 0x00220d89... (77 %)
Panel-0 Writing at 0x00226efa... (79 %)
Panel-0 Writing at 0x0022d6b2... (80 %)
Panel-0 Writing at 0x002332f0... (82 %)
Panel-0 Writing at 0x003104d0... (83 %)
Panel-0 Writing at 0x00318326... (85 %)
Panel-0 Writing at 0x00320622... (86 %)
Panel-0 Writing at 0x003289b7... (88 %)
Panel-0 Writing at 0x00330e4d... (89 %)
Panel-0 Writing at 0x003391d0... (91 %)
Panel-0 Writing at 0x00341676... (92 %)
Panel-0 Writing at 0x00349b03... (94 %)
Panel-0 Writing at 0x00351eda... (95 %)
Panel-0 Writing at 0x0035a453... (97 %)
Panel-0 Writing at 0x0036282b... (98 %)
Panel-0 Writing at 0x0036a9b3... (100 %)
Wrote 4194304 bytes (1099450 compressed) at 0x00000000 in 99.6 seconds (effective 336.7 kbit/s)...
Hash of data verified.

Leaving...
Staying in bootloader.

Panel-0Firmware successfully flashed. Unplug/replug or reset device 
to switch back to normal boot mode.
`
i shutdown and start again 

open candle and only show [CTRL + X]

[image]

but nothing more and dont respond to interaction of controls 
when im trying to writte parameters:

[image]

seems to respond ok but when im reload with another USB port nothing show when refresh button pressed, all values return to 0 after set parameters candle still show only [CTRL + X]

[image]

can please help me to make this work? im quite desperate im totally newbie on this

carlosmorenoaguilera, 2024-03-17:
any suggestion that how i can make a clean install ?

iapx432, 2024-05-04:
Try connecting a different gCode sender to the DLC32 and see what you get returned. There are many alternatives to Candle. Choose one you are familiar with. I use OpenBuilds Control. This will show you the initial dialogue between the sender and the DLC32. This might help you determine if the issue is with Candle or the DLC32.