makerbase-mks/MKS-DLC32 #469: Schematic ou gerber file

greentocic999, 2024-03-31:
Hi Everyone, i need to get a schematic ou gerber file of the mks-dlc32 V1.1 (LS ESP32 PRO V1.1).
Grateful

killthekid, 2024-04-11:
@greentocic999 - did you have success for the file? Currently i'm looking for a shematic of the PCB. Do you have any documents?

greentocic999, 2024-04-11:
I was unsuccessful, in fact. But what is your question, I can try to help you

killthekid, 2024-04-11:
I try to install a Limit switch for Z, as the pcb has a connector for it.
On the custom configuration in fluidnc i can select a (gpio) port, which the limit switch is connected to.
I don't know which port to chose...
Also tried to measure/trace the signal pin on the pcb to the esp, without success.

greentocic999, 2024-04-13:
What's the version of your pcb?

killthekid, 2024-04-14:
It's LS ESP32 V1.0
i thing about going on a MKS LS ESP32 PRO, any recommendations?