makerbase-mks/MKS-DLC32 #474: Please join Discord for private conversation and send me your e-mail. https://discord.gg/BqjUESKq

Charus-OV, 2024-05-11:
Please join Discord for private conversation and send me your e-mail. https://discord.gg/BqjUESKq

_Originally posted by @MJ666 in https://github.com/makerbase-mks/MKS-DLC32/issues/457#issuecomment-2056591357_
Hello friend. Could you send me the firmware by email 4arusov75@gmail.com

closed 2024-05-12 by Charus-OV

andreivenig, 2024-05-13:
Hello, can you help me please with the firmware at andreivenig@gmail.com

Nox-jpg, 2025-02-09:
Hi, I need help setting up some firmware. Could you help me?

Email: greenyfox08@gmail.com

MJ666, 2025-02-10:
The latest firmware released here for the Display is working. Von meinem/meiner Galaxy gesendet

  conversation and send me your e-mail. https://discord.gg/BqjUESKq (Issue #474)