makerbase-mks/MKS-DLC32 #476: MKS DLC32 V2.1 lost my firmware and can't restore it.

Robhoski, 2024-05-13:
I have an MKS DLC v2.1 card and I screwed up the firmware trying to flash it with Grblhal from Open Builds Control. I'm just trying to figure out how to restore the factory settings. I've spent the whole day down the rabbit hole of flashing firmware. I've seen a million folders inside a million more folders with file types of every flavor. I've seen voiceless videos and intalled flashing software of every suggested type. I've attempted to download bin files and install them with various uploaders and followed links to 404 error pages until I wanted to drop kick my computer across the room. Is there any chance any of you could explain in simple terms how to restore the factory firmware on my card so that a caveman like myself can understand it? I was smart enough to save the original grbl settings in Open Builds Control before screwing everything up by trying to use their built in firmware flashing wizard.

Mar4M, 2024-07-03:
had the same problem. have you solved it?

Robhoski, 2024-07-03:
Yes I did find the solution. If you can wait until after the 4th I'll be
home and can help you. Message me then.

Robhoski, 2024-07-16:
> > Yes I did find the solution. If you can wait until after the 4th I'll be home and can help you. Message me then.
> > […](#)
> 
> Do you mind posting your solution here?? Thank you.

Okay - Here's what I remeber. It was pretty easy once I figured out the steps involved. It required flashing the MKS board with a bin file. I put the tools I think I used in a folder on my Google Drive. I know the bin file that5 is sitting alone next to another folder inside the zipped folder is the file you want to flash to the card. It's been a while but I learned the trick by searching YouTube vids about flashing the card. There was one very helpful vid but I can't seem to find it in my history anymore. Anyway, the process is realitively straight forward. You have to use a flashware updating tool to flash the card with the bin file I mentioned. I'm not sure if I used the ESP tool inside the folder but I included it anyway just in case. Maybe search how to flash a card using that tool and the solution will appear. The goal is to load that bin file onto the MKS DLC32 card. I remeber there was a small hiccup that required a slight workaround when I tried it but I don't recall exactly what it was. But with the items in this folder your should be able to start getting close to recovering the board's functionality. The most important file in the folder is the bin file. If you can get that to load onto the card you should be back to the original configuration for that board. Good luck. Hope it works. Here's the link to the folder in my drive that has the files in it. https://drive.google.com/drive/folders/1etDLBrdG-OlMKE1BX8JxwXrAV7cARy-l?usp=sharing

Mar4M, 2024-07-17:
thank you for the response. my present problem of powered axes drift started after i was forced to upgrade from windows 10 to 11 and just after expanding my cnc machine size (have to change and elongated the wirings). i can actually see the z axis motor suddenly turn on randomly - ruining the job. the x and y axes do not reliably return to its origin also. when i press the jog control buttons in UGS for a continuous movement, that moved axis will not stop moving  and the mks board will not respond anymore even if i disconnected my laptop - only a reset or turning it off and on will stop it. did not observed this with candle and gsender (although the haphazard drifts are still observable with these 2). already tried updating the laptop drivers and java to latest versions for windows 11. also deleted and reinstalled the g code senders. but having problems updating the mks v2.1 board.
currently i cannot load the file using the available flashwares as seen in youtube vids. seems like the v2.1 mks board that i have is write protected. was able to flash the v1 board before. after i posted and asked you, noticed that the problem is more severe if im using a bigger spindle (500w vs 200W), doing surfacing jobs and using UGS. somehow after adding some ferrite shielding to the wires and using GSender did not noticed the problems on small projects (<20 minutes duration). but i cannot accept the uncertainty, so i eventually decided to abandon these mks boards. my mach3 and 4axis grbl boards just recently arrived from the mail. hopefully they work.