makerbase-mks/MKS-DLC32 #477: BUGS in last firmware LS ESP32 Pro V 2.1

stef-ladefense, 2024-05-19:
hi,

here is the list of bugs that I found in first 24 hours with this motherboard :

1)
only one LED red alarm works, the others are apparently not managed... or the documentation is not up to date. (page 5/6).
Revision Date：2023-09... 

2)
The Z-endstop does not return the correct information
it returns |Pn:P> instead of |Pn:Z>, so it passes for the probe
the Xendstop returns well |Pn:X>
the Yendstop returns well |Pn:Y>
so home Z ($HZ) does not work (nor $Homing/Cycle0=Z)

3)
with an A4988 Z driver installed, with the switches on, so M1, M2 and M3 at +5v, therefore configure in 16 microsteps.
regardless of the value of $102, the travel of Z is 2mm. always!
(with of course $45=2 (xyz axis mode))

4)
the list of $$ values is not in order, it's all mixed up.
moreover where does "$453=10,000 ;unknown" come from?
many values are misnamed or unknown, from $40 to $54
it does not correspond to page 25 of the documentation

5)
restriction on the wifi key limited to 32 characters instead of 64, and with only letters and numbers! this is not the norm.

it's painful to endure the inconveniences of what is new
and it is even more so to not have the sources available to correct your work since you systematically let your creations die !

i use last firmware : LAS_MKS_ESPS3 PRO V4.1_V1.0.1_20240124
found in (https://github.com/makerbase-mks/MKS-DLC32/tree/main/MKS-ESP32-PRO-main/firmware)

but is this the correct firmware for this card "LS ESP32 PRO" in revision "V2.1_002" ?

SB

stef-ladefense, 2024-05-23:
I found how to adjust the Z steps
it is not the $102 provided for this purpose but $49...
why did you split $102 into 49?

stef-ladefense, 2024-06-05:
for (2) in doc grbl

Input Pin State:

Pn:XYZPDHRS indicates which input pins Grbl has detected as 'triggered'.

Pin state is evaluated every time a status report is generated. All input pin inversions are appropriately applied to determine 'triggered' states.

Each letter of XYZPDHRS denotes a particular 'triggered' input pin.

X Y Z XYZ limit pins, respectively
P the probe pin.
D H R S the door, hold, soft-reset, and cycle-start pins, respectively.
Example: Pn:PZ indicates the probe and z-limit pins are 'triggered'.
Note: A may be added in later versions for an A-axis limit pin.
Assume input pin letters are presented in no particular order.

One or more 'triggered' pin letter(s) will always be present with a Pn: data field.

This data field will not appear if:

It is disabled in the config.h file. No $ mask setting available.
No input pins are detected as triggered.

Nasreddine1991, 2024-06-25:
[image]
الحل

Grex3D, 2024-08-14:
Hi,
I have a issue when I want to use my laser in offline mode. 
When I start engraving via the SD card, the laser jerks (no issues with the same file via USB). I think there might be a problem with the reading speed.

MisogiKumagawa, 2024-11-26:
Hi, i'm setting up a machin with z axis... there are any news on the problem of the Pn:z mistakend with the P output...?

stef-ladefense, 2024-11-26:
of course not!
makerbase doesn't care about its customers, when they pay for their crappy cards, nothing more interests them!
so bug fixes, let me laugh!
it's exactly the same problem with serial screens, fortunately some work with an alternative firmware that has its sources open!

Jacek123458, 2024-11-29:
Totalne gówno !!!

Jacek123458, 2024-11-29:
Nie działa z programem lasergrbl (wersja corexy) nie działa twarde bazowanie. BADZIEWIE nie warte uwagi.
Dokumentacja jakby pijany ją pisał. Zlepek chaotycznych informacji.
Wypuszczają następną płytę nie poprawiwszy błędów w pozostałych.
Zwracam bubel i przechodzę na fluidnc (poradzę sobie bez pracy offline)

MisogiKumagawa, 2024-12-16:
> of course not! makerbase doesn't care about its customers, when they pay for their crappy cards, nothing more interests them! so bug fixes, let me laugh! it's exactly the same problem with serial screens, fortunately some work with an alternative firmware that has its sources open!

oh, can i ask what alternative firmware?

stef-ladefense, 2024-12-19:
> > bien sur que non ! makerbase ne se soucie pas de ses clients, quand ils payent leurs cartes merdiques, plus rien ne les intéresse ! alors corrections de bugs, laissez moi rire ! c'est exactement le même problème avec les écrans série, heureusement certains fonctionnent avec un firmware alternatif qui a ses sources ouvertes !
> 
> oh, puis-je demander quel firmware alternatif ?

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

zombodotcom, 2025-04-07:
Look what I found

```
V4.00(8M.H35.20240226)]
[VER:1.1h.2024022601:]
[OPT:VPH,63,256]
[MSG:Using machine:MKS DLC32]
ok
[MSG:Mode=AP:SSDI=MKS_DLC12457:IP=192.168.4.1:MAC=xxxxxxxxxxxxxxxxxxx
ok
ok
ok
[MSG:Wifi Connecting...]
[MSG:Wifi Connected]
ok
[MSG:Start mDNS with hostname:http://mks_grbl.local/]
[MSG:HTTP Started]
[MSG:TELNET Started 8080]
ok
[MSG:Wifi Connecting...]
[MSG:Wifi Connected]
ok
[MSG:Start mDNS with hostname:http://mks_grbl.local/]
[MSG:HTTP Started]
[MSG:TELNET Started 8080]
192.168.0.78
ok
_connected=1
_connected=1
_connected=1
_connected=1
_connected=1
_connected=1
```

[image]

[image]

https://baizhongyun.cn/home/mkstoolview

works for me!

also used the mks tool in the firmware folder here in this github repo