makerbase-mks/MKS-DLC32 #483: MKS DLC32 v2.1 + TMC2209 drivers issue

makrosloro, 2024-06-19:
Hi all!

I have the following setup in my laser cutter/engraver project:

- MKS DLC 32 v2.1
- TMC 2209 Driver
- 3 Nema 17 stepper motors

I have tried various configurations without success; the motors do not work correctly at any speed (low - mid - fast). When I try to run them with the screen, they jump.

Additionally, the motors are only turning in one direction.

[image]
[image]

Right now, this is the configuration I have set up (default one from this github repo):

```
/***************************************************************************/
DLC_TEST=0        //Model： 0：Usage mode， 1：Test mode
/********************************************************************************/

/***************************************************************************/
*--------------Machine parameter format："$x=y"
*--------------Function configuration format："[ESPXXX]yyy" 
*--!!!WARNING!!!--*
*------1.Do not change the value of x in the machine parameter format at will, 
         otherwise a configuration error will occur and it will not start
*------2.Can't add configurable items at will
/*******************************************************************************/ 
// (Step pulse time：Sets time length per step. Minimum 3usec.)
$0=10 	
// (Step idle delay:Sets a short hold delay when stopping 
// to let dynamics settle before disabling steppers.
// Value 255 keeps motors enabled with no delay.)		
$1=5 		
//---Basic parameter setting-----//
//(Step pulse invert) 
$2=0 			
//(Step direction invert)
$3=0 			
//(Invert step enable pin)  
$4=0 
//(Invert limit pins) 			
$5=1 	
//(Invert probe pin) 		
$6=0 
//(Status report options)			
$10=1 
//(Junction deviation) 			
$11=0.010 	
//(Arc tolerance)   	
$12=0.002
//(Report in inches) 		
$13=0 
//(Soft limits enable)			
$20=0 
//(Hard limits enable)			
$21=1 		
	
//---Zero setting----//
/(Homing cycle enable)
$22=1 
//(Homing direction invert)			
$23=1 	
//(Homing locate feed rate) 		
$24=300.000
//(Homing search seek rate)  	
$25=1000.000
//(Homing switch debounce delay)	
$26=250.000
//(Homing switch pull-off distance)  	
$27=1.000 	
//(Maximum spindle speed)	
$30=1000.000 
//(Minimum spindle speed) 	
$31=0.000

//---Protection function setting----//
//(Laser-mode enable)	
$32=1
//(Flame  1:Turn on,0:close) 
$37=0 
//(Beep 1:Turn on,0:close)  
$38=0 
//(Gyro 1:Turn on,0:close)      	
$39=0
//(Set Language, 0: ch, 1:en 2:de)
$40=1
//(Roller 1:Turn on,0:close) 	
$41=0

//(X-axis travel resolution)
$100=80.000
//(Y-axis travel resolution)  	
$101=80.000 
//(Z-axis travel resolution) 	
$102=80.000 
	
//---Speed setting----//
//(X-axis maximum rate)
$110=6000.000 	
//(Y-axis maximum rate)
$111=6000.000
//(Z-axis maximum rate) 	
$112=6000.000 	
//(X-axis acceleration) 
$120=500.000 
//(Y-axis acceleration) 	
$121=500.000 
//(Z-axis acceleration)	
$122=500.000 
//(X-axis maximum travel)   	
$130=300.000 
//(Y-axis maximum travel) 	
$131=300.000
//(Z-axis maximum travel)  	
$132=80.000 	

//--WIFI configuration---//
[ESP110]STA
[ESP131]8080	
//wifi name
[ESP100]MAKERBASE3D 	
//wifi password，：[ESP101]makerbase3d
[ESP101]makerbase3d	

/*---modify2021-11---*/
```

makrosloro, 2024-06-23:
Anyone can help with this?

zebsboards, 2024-06-27:
Have you tried setting this to 255 to remove the idle delay?

// (Step idle delay:Sets a short hold delay when stopping 
// to let dynamics settle before disabling steppers.
// **Value 255 keeps motors enabled with no delay.)**		
$1=255

makrosloro, 2024-06-29:
Gonna try and reply. Thank you in advance.

zebsboards, 2024-06-29:
Just put mine online last night using 2209 drivers and it worked out of the box using the preinstalled firmware and cfg file.I did have to invert the limit switch entry as I’m using common ground limit switches.I’m not using the original motors though, I’m using 1.8 degree steppers and even the steps preset was bang on.Just using laserGRBL for testing if that helps.

makrosloro, 2024-07-04:
Still not working. I add some aditional info. My stepper motors are 17HS4401 model.

zebsboards, 2024-07-04:
Sounds to me like you haven't got the motors wired correctly.
Here's a link with the wiring pinout including some colour codes .... https://www.aliexpress.com/item/1005005128339128.html?src=google&src=google&albch=shopping&acnt=631-313-3945&slnk=&plac=&mtctp=&albbt=Google_7_shopping&gclsrc=aw.ds&albagn=888888&isSmbAutoCall=false&needSmbHouyi=false&src=google&albch=shopping&acnt=631-313-3945&slnk=&plac=&mtctp=&albbt=Google_7_shopping&gclsrc=aw.ds&albagn=888888&ds_e_adid=&ds_e_matchtype=&ds_e_device=c&ds_e_network=x&ds_e_product_group_id=&ds_e_product_id=en1005005128339128&ds_e_product_merchant_id=109096494&ds_e_product_country=CA&ds_e_product_language=en&ds_e_product_channel=online&ds_e_product_store_id=&ds_url_v=2&albcp=19366866438&albag=&isSmbAutoCall=false&needSmbHouyi=false&gad_source=1&afSmartRedirect=n

if you can't match the colour codes, use  a multimeter and measure out the wire pairs.  A pair that has no continuity at all is not a coil pair, it's one of each coil.  A pair that measures out to 10 ohms or less is a pair (A+/ A-, B+ / B-)
The minus and plus don't matter to get it moving, they just determine which direction it goes on the command and you can either rewire one pair (reverse the pins) or invert the step/direction signals in the cfg settings to match it up.

If you are currently wired up with one wire from each coil as a pair, the motor will bark and stall and fight itself trying to move on command.

makrosloro, 2024-07-05:
Well, after several tries I noticed that stepper motor wires are not the good ones. I've changed it and now it's working! Only one more thing, y+ and y- screen buttons make Y axis go in only one direction. How can it be solved?

Thank you for all your help and patience!

zebsboards, 2024-07-05:
It sounds like you have 1 pair of wires connected in reverse to the other.

example:  A+ to A+, A- to A- , B+ to B-, B- to B+

pick one pair of wires and reverse them.

If the axis runs backwards after you reverse the wires you can either put it back to how it was and reverse the other pair (hardware fix) or go into the cfg file and invert the step and direction signals for that axis (software fix).

RC0D3, 2024-08-24:
I have trouble with mine changing the main board.
If you seed at [PDF from manual wiring](https://github.com/makerbase-mks/MKS-DLC32/blob/main/MKS-DLC32-main/doc/DLC32%20wiring%20manual.pdf) page 7, wiring from PCB is different from TMC2209 datasheet.

Board order:
1B
1A
2A
2B

TMC2209:
2A
1A
1B
2B

I will figure out to solve that and return...

RC0D3, 2024-08-25:
I got that working now, changed pins, but now I know that I'm using wrong firmware of my laser, then 2 axes run at same time

rana-ans, 2025-02-06:
> \o/ changed firmware and inverse one step motor connector (inverted it). Now it's working. I change my connector like this [video](https://www.youtube.com/watch?v=8f55FCvBYMo) used a clip to do and a solid base (be careful).
> 
> I used a lot of things, multimeter, battery to test correct motor way, etc etc etc.

can you tell us what firmware did you use and the whole process where you fixed both axes running at the same time even if you move either axes? thanks.