makerbase-mks/MKS-DLC32 #484: Need advice wiring and using MKS DLC32 v2.1 with 100w CO2 Laser PSU.

Arion1936, 2024-06-22:
Hi to all! I would like to ask for your kind help.
I have purchased the board MKS DLC32 v2.1 and I am planning to use it with Lightburn to control a standalone XY CO2 Laser machine.
I have searched everything that is available on the internet, but I feel very confused about the proper route, as it seems that there are many different wiring configurations. I would like not to try them all risking my safety and equipment.

This is the equipment:
-Laser PSU MYJG100W with external LCD potentiometer
-80W CO2 Laser Tube
-MKS DLC32 v2.1 with TFT35
-Additional 24V external PSU
-I have not decided yet about Nemas 23 stepper motor drivers. Will use external drivers.

The main confusion is how to wire properly the board with the LPSU so:
1. Prevent the tube to fire on the startup, when powering the board. This is an issue that many have.
2. Been able to use the software to control the PWM when cutting and engraving.
3. Been able to test fire the LPSU without the software.
4. Been able to use safety switches and laser tube enable ON/OFF switch.

The LPSU has the following connectors: H, L, P, G, IN, 5V.

Has anyone tried something like this and actually make it to work?
I would really appreciate any advice and help!

Thank you all in advance!

Benglin, 2024-07-27:
Hi @Arion1936 I didn't see your post and just made one for [my scenario](https://github.com/makerbase-mks/MKS-DLC32/issues/493), I'm wondering if you have resolved your connection issue? If so I can definitely learn something from you; if not, I will update you if I resolve it (and update my post).

I saw a very popular post [here](https://github.com/makerbase-mks/MKS-DLC32/issues/48) that got few working suggestions there (just that my scenario is with weird TTL/PWM voltage), perhaps you can find your solution there too 🙂

Benglin, 2024-07-30:
Hi @Arion1936, just so you know, I have solved the wiring issues in [my scenario](https://github.com/makerbase-mks/MKS-DLC32/issues/493) in case that helps you too.

mohoxo, 2025-01-09:
[image]
This is the latest connection method, which can control the CO2 laser. The firmware still needs to be modified,