makerbase-mks/MKS-DLC32 #490: WOW repositories are empty and firmware is ummmmmm

jnl1105, 2024-07-09:
I received the MKS DLC32 today...... The display was showing backwards and kind of jumbled so I tried flashing the firmware as suggested on the github....wow it has ended up being a f**kangular (#when you are f**ked from multiple angles LOL) experience. The main page firmware links go 404. The tool to upload the firmware rar is bad. (I tried using yours and the supplied firmware) the supplied firmware is bad, no display nor able to connect to a grbl controller. I did try a sculpfun one and it loaded and was able to be controlled through grbl. There was no display though. So Now I am trying to find a decent firmware that works with the MKS DLC32 V2.1. LOL wow its been fun

stef-ladefense, 2024-07-10:
welcome to the world of MKS who do not track their products

jnl1105, 2024-07-10:
First as my excuse I was recently diagnosed with neurological pathway issues that sometimes make it hard for me to process information and grasp visual or audible cues...... I wasn't downloading the firmware correctly I was right clicking and selecting "save link as" Instead of clicking on raw.... I should know I have even had projects on github. So I big dumb dumb head

closed 2024-07-10 by jnl1105