makerbase-mks/MKS-DLC32 #493: Need Help: Connecting MKS DLC32 v2.1 to CO2 Laser PSU

Benglin, 2024-07-26:
Hi MKS team and community experts, I will try to make this post as clear as I possibly can so you can help me out. These might be beginner questions so please bear with me... 🙂

## My MKS DLC32 v2.1

Few points about my **MKS DLC32 v2.1**:

-   It is powered by 12V DC
-   I have never flashed any new firmware so it's the default
-   I am afraid to connect it to CO2 laser PSU due to the observation below about signal voltage

-   ❓ **Question 1**: why is the output for `S` in `S, TTL, V` measured to be `0 ~ 12V`? (I know we can't quite measure PWM signals with a digital multi-meter, but its value should not have gone way beyond `5.00V` if it is meant to be within `0 ~ 5V`).

The [document](https://github.com/makerbase-mks/MKS-DLC32/blob/main/MKS-DLC32-main/doc/DLC32%20wiring%20manual.pdf) says it should be `0 ~ 5V` as shown below:

[image]

-   ❓ **Question 2**: should I use a step-down mechanism to bring it down from `0 ~ 12V` to `0 ~ 5V`?

-   ❓ **Question 3**: would refreshing with the latest firmware fix the issue (I can try that when I return to it next week)?

## My CO2 Laser PSU

Not wanting to destroy my [laser PSU](https://www.aliexpress.com/i/32818973603.html), I made my [Arduino Uno](https://store-usa.arduino.cc/products/arduino-uno-rev3?srsltid=AfmBOopZtP1FUiXeNEp1uW28MLvQs69g982alkaQwaMi8ZoRNANcoRfl&selectedStore=us) output PWM signals of `0 ~ 5V` through a PWM pin `9`, here's how I connect the wires:

[image]

-   ❓ **Question 4**: I am sure the PWM signal is `0 ~ 5V` (using a digital multi-meter, and an associated LED that fades in and out)

-   ❓ **Question 5**: Should I even be using that `5V` output on the laser PSU? It seems to be an _"output port"_

-   ❓ **Question 6** Is it because I did not connect `WP` to `G`? If so I'm gonna be so pissed with myself now

The above connection is inspired by the following wiring diagram that has been shared in various places before (I just replaced the potentiometer with PWM that's coming from an Arduino Uno, it will be changed to MKS DLC32 v2.1 when wiring confusion is sort out)

[image]

Thank you all for spending the time to read this, any help I get would really be appreciated. Thanks again team 😄

costycnc, 2024-07-27:
https://www.youtube.com/shorts/79jmSSWvY7w

Benglin, 2024-07-27:
Thanks for the tip, @costycnc! That looks neat 👍

I see it uses `SPINDLE (-)` to drive the PSU, which is something I might want to try out too (based on the [documentation](https://github.com/makerbase-mks/MKS-DLC32/blob/main/MKS-DLC32-main/doc/DLC32%20wiring%20manual.pdf) it should produce `0 ~ 12V` signal for me, I'm not sure if my LPSU can handle that, I'll check with Cloudray).

The only thing I can't tell from the video is which port does the `SPINDLE (-)` connect to in the LPSU (as K40 one is quite different from [mine](https://www.aliexpress.com/i/32818973603.html)). But thanks anyway for your help!

costycnc, 2024-07-28:
i have inspired from this video https://youtu.be/xfqtSgPIYKg?si=ajWVxvMtEqqSMOA0&t=382

Benglin, 2024-07-30:
Thanks again for your help, @costycnc! Your video prompted me to measure the spindle's voltage (unfortunately that is also ranging from `0 ~ 12V` when my laser PSU expects an input signal of `0 ~ 5V`), then re-measure the power output of the `S-TTL-V` ports. It turns out that, unlike my original claim in my question above, the `S` pin does output `0 ~ 5V` PWM signals! I think my previous measurements were wrong due to some bad wiring.

In any case, it turns out that I did not connect `P` (water protect) to `GND` on the laser PSU. Since the water protection is not a closed circuit, the laser PSU refuses to fire the tube. After connecting `P` to `GND`, everything works perfectly!

The diagram below illustrates the correct way to connect my laser PSU to the MKS DLC32 board (note that the pins on MKS DLC32 are incorrectly labeled as `S-TTL-V` when they should really be `S-GND-V`):

[image]

And to answer my own questions in the original post:

-   💡 **Answer 1**: The `S-TTL-V` pins on the MKS DLC32 board are wrongly labeled; they should really be `S-GND-V`. My previous readings on these pins were wrong. The `S` (signal) pin correctly outputs `0 ~ 5V` as documented, and `V` outputs a constant `12V` or `24V` depending on the input voltage to the MKS DLC32 board.

-   💡 **Answer 2**: Since the `S` pin correctly outputs `0 ~ 5V` PWM signals, no step-down is required.

-   💡 **Answer 3**: I didn't get to try the latest firmware, and I don't have a reason to do so right now, so this remains a mystery.

-   💡 **Answer 4**: The `5V` on the laser PSU actually outputs a constant `5V`, and there's no use for it in my scenario.

-   💡 **Answer 5**: This is exactly the reason. I did not connect `P` (water protect) to `GND`, causing the laser PSU to "think" that the water protection system is "broken," therefore refusing to fire the laser tube.

closed 2024-07-30 by Benglin

jlbarros76, 2024-08-12:
Hi Benglin, what a wonderful work . I have a 150W PSU and 100W CO2 Laser tube, can I use this connection (the last one) without problem? I´m stuck finding a solution to make functional my diy cnc laser engraver. Can I use with laser grbl o lightburn? Waiting your comments, thank you.

Benglin, 2024-09-02:
Hi @jlbarros76, I'm sorry for not seeing your question earlier! I am not certain of what the issue you are facing with, can you explain a little more and also show us how the connection is made? I might not be able to help you (because I am a beginner also) but others should be able to 🙂

By the way, I am having an issue myself also, the final connection works, but I realized the laser is firing even when I just turned on the laser PSU. I am not sure what is the problem here but I will continue to debug it (because it did work before with my Laser Web).

costycnc, 2025-10-26:
perhaps is more explicit
https://www.youtube.com/watch?v=QZNbPhGoNkQ
https://www.instructables.com/Upgrade-K40-Laser-With-Mks-dlc32-Makerbase-Wifi/
https://www.costycnc.it
[image]