makerbase-mks/MKS-DLC32 #495: Unable to update with any firmware

steersteer, 2024-08-05:
Hi,
I'm now on 6 hours attempting to get this board to do anything.
I initially started with the WiFi set-up, it appeared to work but nothing showed up on my network.
I decided to get the latest firmware and this is all that happens:
[image]

Any ideas?

meistertek, 2024-08-08:
> Hi, I'm now on 6 hours attempting to get this board to do anything. I initially started with the WiFi set-up, it appeared to work but nothing showed up on my network. I decided to get the latest firmware and this is all that happens: 
> 
> Any ideas?

@steersteer This messed me up for quite a few hours as well! There is a mistake in the document that guides you through the process of using the MKS Firmware Tool, the ESP32 chip is not standard and has 8MB of flash memory. If you select FLASH SIZE: 8MB (or 16MB) the firmware loads into the board and also the touchscreen works instead of being blank. I figured this out by viewing this helpful youtube video by Mani Fa: [MKS DLC32 and TS35-R](https://youtu.be/Qz-weOHobjk?feature=shared)

Here is the screen shot that shows the correct setting for the flash tool:
[image]  

Hope this helps!

steersteer, 2024-08-09:
Thanks for that. 
It didn't initially work for me but I changed usb cable and bingo! I'm all up and running now.
Just have to work out my fluidNC issues now

JSF9026, 2024-10-13:
can you please share the bin file or the link to download it ?

closed 2024-10-13 by steersteer