makerbase-mks/MKS-DLC32 #496: TTL max tension for 100% Laser

Marneus73, 2024-08-06:
Hello,

I have some issues with my laser, so i decide to check the TTL tension for 100% laser.
I measure with a multimeter directly on the card (TTL PIN) between signal and ground : 4.88V, not 5V.

Has anyone ever noticed this ? Any solutions ? Is it a firmware (config) or hard problem ?

Thanks in advance.

stef-ladefense, 2024-08-20:
4.88 is within TTL tolerances.
moreover the power is managed with the duty cycle and not in voltage.

how is your laser connected? is it a 3 wire?
what model of laser? of card?

Marneus73, 2024-08-20:
Hello,
It' a 4 wire laser head PLH3D-XT-50 
- Alimentation 12V (plug directly on the power supply)
-  TTL signal with it's own ground (both plug on the DLC32)

My card is a MKS DLC32 V2.1_003