makerbase-mks/MKS-DLC32 #498: Request to retain the user X~Y origin if a "sculpture" is paused and stopped prior completion.

stev-e, 2024-08-14:
Hello.
My board is driving a TwoTrees TTC-450 hobby router.

I can set a "user" origin before starting a gcode file.

If the file runs through to completion the origin values are retained and it homes as expected.
If I stop the file before completion and go to the "movement" screen the x and y values shown bear no relation to the toolhead position, and pressing the home button makes it move to a random position, the user origin has been lost.

Can the firmware please be adjusted so that the user origin is retained ?

Thank you.

Luimohr, 2025-03-25:
Olá, realmente esse problema existe, gostaria que fosse resolvido também

Luimohr, 2025-07-28:
Olá, alguém mais está com esse problema? Alguém teria a solução?

Luimohr, 2026-04-05:
Alguem ja consegui resolver esse problema de perder a posição ao interomper a operação!