makerbase-mks/MKS-DLC32 #505: MKS DLC32 - Firmware file

JSF9026, 2024-10-13:
Hello,

I have recently purchase MKS DLC32 board as an upgrade following the MKS DLC basic version.

Currently i was trying to find the firmaware file .bin to flash the firmware for my CNC machine, but the links to GitHub location is broken and giving error.

Can anyone share the latest firmware file ?

Thanks

bottleworks, 2024-10-15:
https://github.com/makerbase-mks/MKS-DLC32/tree/main/MKS-DLC32-main/firmware/TS35/Laser/Normal/Board_V2.0
This is the section for the laser based firmware.

JSF9026, 2024-10-15:
Thank you for your reply and link.

does it also work as normal cnc router with spindle ? Or does it need a different bin file ? 

Thanks