makerbase-mks/MKS-DLC32 #506: Where is the source code living these days?

factoidforrest, 2024-11-01:
I see a repo with some source with very little docs that hasn't been updated in two years. I do not think that is the mainline repo anymore. I want to change the baud rate and recompile the firmware for my 2.1 board. 

https://github.com/makerbase-mks/MKS-DLC32-FIRMWARE

Is this the most up to date firmware repo? It seems very unlikely.

Also, I'd love to fix the issue where unsecured wifi networks do not appear to be supported.

dosAnjosGustavo, 2024-11-04:
please guys, share the latest source code with community @makerbase-mks @solawc @elf128 @abinoer @mks-viva @MS1987 @nehilo @KevinOConnor

terjeio, 2024-12-21:
> please guys, share the latest source code with community

They are obliged to publish the souce code. Most likely stolen from FluidNC.