makerbase-mks/MKS-DLC32 #507: Z Axis control on the screen

Toaro, 2024-11-09:
The last firmware version 2.30 for laser CoreXY MKS DLC32 lost the Z control axis on the TS35 Display, wich are present in the 2.1 version

Can you add these controls on the next version, or correct the 2.30? Thanks

ok1408, 2025-01-13:
I would like to urgently join this request and am also waiting for the moveable z-axis for my laser with the MKS TS35 touchscreen.
Thanks in advance :-)

mvvproject, 2025-06-08:
Hi, I need to use the Z axis to control the height of the laser above the material, but the firmware does not have this feature, how can I fix this?
[https://github.com/makerbase-mks/MKS-DLC32/tree/main/MKS-DLC32-main/firmware/TS35/Laser/Normal/Board_V2.0](url)

If this is difficult for the author, can he open the firmware source code for its modification by users of this board? Otherwise, the board is useless to use.