makerbase-mks/MKS-DLC32 #515: MKS DLC32 MAX - Values for $101 and $102 cannot be changed

psy0p, 2025-01-04:
If I try to change '$101' or '$102', the controller crashes and restarts. It seems that the two values ​​are duplicated for some reason. If I query the parameters with '$$', it shows these two values ​​twice. Once as '80' and once as '80,000' (see photo). The firmware is freshly flashed and I have not changed anything. How can I fix this problem? Thanks for your help...

[image]

[image]

cr8zyolerece, 2025-01-20:
I'm having the same issue.  Is there a way to edit those two duplicate $101 & $102?  Also there seems to be an issue here with $51, $52, $53, & $54 as well.

[image]

psy0p, 2025-01-21:
The board still has a lot of errors in the firmware :-/ 
I was able to get around the problem with $51 -$54 like this:

$Wifi STA PWD=YourPassword
$Wifi STA SSID=YourSSID
$Wifi AP SSID=MKS_DLC
$Wifi AP PWD=12345678

 ...and you can change $101 + $102 like this:

$Y/StepsPerMm=200.000
$Z/StepsPerMm=200.000

It will still be displayed twice, but the new values will be adopted and then it will work.

@Makerbase: Please release the source code if you are not willing to fix the errors yourself.

cr8zyolerece, 2025-01-21:
Wow! Thanks bro!  This is all I needed. I’m sure you guys will work out the other kinks! Have a fantastic day sir. Maurice

oLuisMatias, 2025-01-24:
Thank you. It works :)

DavidHolecko, 2025-02-15:
$Z/StepsPerMm=25.000  The value changes, but when manually moving in LightBurn, it still operates at the preset value. :( 

[image]

oLuisMatias, 2025-02-16:
Install new firmware. The bug to change value is fixed and maybe you fix your problem.