makerbase-mks/MKS-DLC32 #522: MKS DLC 32 Max

oLuisMatias, 2025-02-04:
Is it possible to have limit switch as a Z probe?
I have X, Y and Z limit switches working properly.

Probe is enabled:
`$Probe/Enable=1`

When I press the probe for some seconds I get the error:
```
error: Probe is triggered
[MSG:Probe err!Please restart the machine!]

```

When I do G38.2 Z-5 F200 i get this error:|

```
G38.2 Z-5 F200
[MSG:No probe pin defined]
error: Unsupported GCode command
```

Anyone can help?

oLuisMatias, 2025-02-05:
G38 working with new firmware version.

Thanks!

closed 2025-02-05 by oLuisMatias

hemanthhr139, 2025-09-20:
Hello, please help me with a similar case. I am using the MKS DLC32 v2.0 for a laser cutting and engraving machine, and I am planning to use a Z-axis auto-focus for the laser module. Will this work with the new firmware?