makerbase-mks/MKS-DLC32 #527: Homing MKS DLC32 MAX

kpc29160, 2025-02-12:
MKS DLC 32 MAX
[V4.00(8M.H35.20250113)]
[VER:1.0.10,20250113]
[OPT:VPH,63,256]

Le homing ne fonctionne pas, que ce soit au démarrage de lightburn, lorsque l'on clique sur le bouton de commande ou lors de la commande manuel $H X ou $H Y.

Les moteurs ne font aucun mouvement pour faire la recherche des fins de courses.
-------------------------------------
Homing does not work, whether when starting lightburn, when clicking on the command button or when manually commanding $H X or $H Y.

The motors do not make any movement to search for the end of travel.

config :
$21=1
$22=1

$24=300.000
$25=3000.000

kpc29160, 2025-02-13:
Problème résolu en envoyant la commande

$Homing/Cycle0= XY
$Homing/Cycle1= Z

betuss2345, 2025-05-07:
Hello,

I have an MKS DLC32 MAX V1.0 board, and it appears to use the ESP32-S3 chip. I would like to use this board with FluidNC or Grbl_ESP32 to control my CNC machine, including probe functionality.

Unfortunately, it seems that the ESP32-S3 is not currently supported by either firmware. FluidNC developers advised me to contact MKS directly for support.

Is there any firmware available (even experimental) for the ESP32-S3 version of this board?  
Alternatively, is it possible to replace the ESP32-S3 module with a standard ESP32-WROOM module to make the board compatible?

Thank you for your support.

Hostaler, 2025-09-20:
FuidNc ya lo tiene. 
https://discord.com/channels/780079161460916227/1416056429185601536