makerbase-mks/MKS-DLC32 #532: CoreXY for the Max

HorizonCDR, 2025-03-01:
I cant find the CoreXY firmware for the MKS-DLC32-MAX
Is it available?

nobimoto1, 2025-05-16:
hello
I have the same problem, kann you help us?

makip69, 2025-05-23:
Hi, 
I am installing the firmware available for the MKS ESP32 PRO board; there is a specific one for core XY. I will start the tests. The hardware is basically the same, so I don't think there will be any problems. I will keep you updated.

DumpsterDave, 2025-12-29:
I too am using a Max for a CoreXY project.  I just stumbled upon FluidNC and was able to easily flash my max with the 4.0 firmware  using https://installer.fluidnc.com.  Once flashed, you can use the same tool to upload and then activate your config from flash.  Just replace the `cartesian:` line under `kinematics:` with `corexy:` and you're off to the races.

Config Example:
http://wiki.fluidnc.com/en/hardware/3rd-party/MKS-DCL32-MAX