makerbase-mks/MKS-DLC32 #533: set vref for dual y axis steppers

mugginsjm, 2025-03-03:
Hi 
I'm using TMC2208 drivers for my Nema17 steppers and have set Vref to 1.2V on the x axis. Because the y driver is controlling 2 steppers do I need to double this to 2.2V ?
thanks

ellensp, 2025-04-07:
Yes. when stepper motors are in parallel, as long as you don't exceed max current of the driver.

betuss2345, 2025-05-08:
Hello,

I have an MKS DLC32 MAX V1.0 board, which appears to use an ESP32-S3 module. I would like to use this board to control my CNC machine, including the probe function, using either FluidNC or Grbl_ESP32.

However, it seems that ESP32-S3 is not fully supported by either firmware at this time. FluidNC developers suggested I contact MKS directly for support.

So I have two simple questions:

Is there any firmware available (even experimental) that works with the ESP32-S3 version of the MKS DLC32 MAX V1.0 and supports CNC and probing features?

If not, can I replace the ESP32-S3 module with a standard ESP32-WROOM module to make the board compatible with FluidNC or Grbl_ESP32?

Thank you very much for your help!