makerbase-mks/MKS-DLC32 #537: Customise boot screen

suganth1999, 2025-03-16:
I need to custom my MKS DLC32 v2 TS35 CNC display please let me know its urgent 

Thanks 🙂

ellensp, 2025-04-07:
On these devices the TFT is controlled by the controller firmware, every single byte and pixel.. There are no asset files to update like you may have seen in serial displays.

On the ts35 it displays Makerbase as the boot screen

[image]

This is the file that contains the data for the Makerbase logo
https://github.com/makerbase-mks/MKS-DLC32-FIRMWARE/blob/main/Firmware/Grbl_Esp32/src/lv_pic/mks_logo.c

you need to build a new version of this file, you only need to update the "#if LV_COLOR_DEPTH == 16 && LV_COLOR_16_SWAP == 0" section as that what the display is using, the file seems to be be auto generated for multiple formats.

You then need to build and upload your new firmware.

nofuturekid, 2025-04-07:
Instead of modifying code, one could also just upload the logo via the WebUI. Of course the format needs to be respected.

ellensp, 2025-04-07:
There are no such settings in my web user interface.  But I have the TS24, and they haven't released that firmware.. So it might be different, but I doubt it.

ellensp, 2025-04-07:
Ah, your talking about https://github.com/makerbase-mks/MKS-DLC32/issues/390