makerbase-mks/MKS-DLC32 #538: Is the MKS DLC32 to slow?

ojjeCNC, 2025-03-23:
Just installed an MKS DLC32 on a 130W CO2 laser.  
On the left, I have 20% power and a speed of 20 mm/sec.  
On the right, I have 20% power and a speed of 100 mm/sec.  

I don’t understand why the laser doesn’t burn all the lines when running at a higher speed.  
When I run the same job on my K40, the result is perfect at the higher speed.  

Could it be that the DLC32 isn’t fast enough?

[image]

Gzikic, 2025-03-23:
Hi, did you set power through mks board or manually?

ojjeCNC, 2025-03-23:
Power is controlled by LightBurn via MKS DLC32.

Gzikic, 2025-03-23:
What schematic do you use?

ojjeCNC, 2025-03-23:
> What schematic do you use?

How do you mean?

closed 2025-03-24 by ojjeCNC

Gzikic, 2025-03-24:
> > What schematic do you use?
> 
> How do you mean?

I build laser with mks dlc32 board, make frame connect motors, now I need to connect power supply for laser tube.
What schematic do you use for that?

ojjeCNC, 2025-03-24:
> > > What schematic do you use?
> > 

[image]