makerbase-mks/MKS-DLC32 #541: Probe reporting Flame detection

Aepx, 2025-03-25:
Hello, I want to use a Z probe but when it is triggered it reports a flame detection.
I don't have or want a flame detector.
How can I use a probe on the probe pins?
I'm using firmware V2.30_H35_2022_1223_N_ZX_001

Aepx, 2025-03-27:
Found a newer firmware that is working.

Walterstucco, 2025-04-17:
> Found a newer firmware that is working.
@Aepx Which one? Does the I2C port works too?

Aepx, 2025-04-17:
I used the 2023 version. I haven't tried the I2C so can't say for sure.
Actually switched to GRBLHAL and everything works.

francoissouza, 2025-04-24:
@Walterstucco I accidentally tested this feature today and I could perform Z probing without problem 😬 

I'm using the V2.30_H35_2022_1223_N_ZX_001.bin laser firmware on a DLC32 and gSender. I know that it does make sense, but I was testing out some features. 

In newer firmware versions, like that you reported, the probe pins are used for the flame sensor. However, **it seems** (I've no verifyed this info in firmware code) that there is a great debouncing time (around 1 or 2 seconds) for this input, and then, if your probing routine does not keep the sensor activated for longer than this, it will not trigger as a flame sensor, but still working for probing prupose. 

It was a clever trick implemented by the MKS team (if it was intentional, of course).