makerbase-mks/MKS-DLC32 #542: Z not homing with Home button

Aepx, 2025-03-27:
I use Lightburn and have it set to Home on connect but the Z doesn't home.
I have a macro setup to home the Z but I would rather it just home when I press the Home button.
Is there a firmware, or just a setting, to do this?

Luimohr, 2025-03-29:
alguém tem a solução?

ellensp, 2025-04-07:
"About CNC Firmware Z-Homing

The command to turn on this feature: ($Homing/Cycle1=Z)or($Homing/Cycle0=Z;$Homing/Cycle1=XY)
The command to turn off this feature: $Homing/Cycle1=0"

This should work with laser too

francoissouza, 2025-04-24:
@Luimohr  and @Aepx, I've tried the solution offered by @ellensp and it works fine! Thanks.
I just need to invert the Z-axis homing direction using the $23.

I think that this issue is closed, but I can't close this by myself.

Luimohr, 2025-04-25:
Olá, para min o eixo Z não movimenta para procurar o Home.
Deve ser esses parâmetros ($Homing/Cycle1=Z) ou ($Homing/Cycle0=Z; $Homing/Cycle1=XY) Porem não sei alterar isso no código fonte, quando altero e gravo na placa o LCD não funciona mais...
Se alguém do fórum puder criar o firmware e postar seria ótimo.
Meu LCD é TS35-RV2.0

francoissouza, 2025-04-25:
@Luimohr, é bem simples alterar isso via comunicação serial. Veja:

Se você usa o Lightburn, após conectar com sua máquina, envie a ela o comando pelo console (minha interface é um pouco modificada, mas a ideia é a mesma):

[image]

Se usas o LaserGRBL, faça o mesmo:

[image]

closed 2025-04-25 by Aepx

Luimohr, 2025-04-28:
Olá, francoissouza, funcionou perfeitamente, muito obrigado