makerbase-mks/MKS-DLC32 #553: MKS DLC32 MAX V1.0 with ESP32-S3 – CNC + Probe Firmware?

betuss2345, 2025-05-08:
Hello,

I have an MKS DLC32 MAX V1.0 board equipped with an ESP32-S3 module. I'm aiming to control my CNC machine, including the probe functionality, using either FluidNC or Grbl_ESP32 firmware.

However, it appears that ESP32-S3 is not currently supported by either firmware. The FluidNC developers suggested reaching out to MKS directly for support.

Therefore, I have two questions:

Is there any firmware available (even experimental) that supports the ESP32-S3 version of the MKS DLC32 MAX V1.0, including CNC and probing features?

If not, is it feasible to replace the ESP32-S3 module with a standard ESP32-WROOM module to make the board compatible with FluidNC or Grbl_ESP32?

Thank you very much for your assistance!

betuss2345, 2025-05-08:
Hello,

I have an MKS DLC32 MAX V1.0 board equipped with an ESP32-S3 module. I'm aiming to control my CNC machine, including the probe functionality, using either FluidNC or Grbl_ESP32 firmware.

However, it appears that ESP32-S3 is not currently supported by either firmware. The FluidNC developers suggested reaching out to MKS directly for support.

Therefore, I have two questions:

Is there any firmware available (even experimental) that supports the ESP32-S3 version of the MKS DLC32 MAX V1.0, including CNC and probing features?

If not, is it feasible to replace the ESP32-S3 module with a standard ESP32-WROOM module to make the board compatible with FluidNC or Grbl_ESP32?

Thank you very much for your assistance!

josepatron, 2025-09-03:
mks dlc32max pins map
[image]